gnuradio/gr-bokehgui #4: localhost:5006/..... works but not 192.168.1.xxx:5006/....

kurtulmehtap, 2017-07-05:
Trying to run the example from: http://kartikpatel.in/GSoC2017/2017/06/23/FreqSink1.html

The address localhost:5006/..... works perfect on my computer,
However when I try 192.168.1.xxx:5006/.... from another computer on the same network the spectrum is not displayed.(The page name is displayed correctly)

kartikp1995, 2017-07-05:
Try to add `allow_websocket_origin=['*']` as an argument to `Server()` constructor in `main()` function of your script. I have updated examples in `examples` folder. 

You will have to pull recent changes and re-build your GNURadio from `bokeh_in_gnuradio` branch from `kartikp1995/gnuradio` in order to build the script from GRC.

Let me know if this solves your issue.

kartikp1995, 2017-07-23:
Closing this assuming issue is solved.

closed 2017-07-23 by kartikp1995

kurtulmehtap, 2017-07-24:
Thanks a lot iti is solved.