gnuradio/gr-bokehgui #5: scripts/start-server.sh missing?

777arc, 2017-09-02:
I'm getting an error at bokehgui.utils.create_server() and it looks like it's trying to run /scripts/start-server.sh which I don't see in the scripts directory (I only see check-port.sh).

kartikp1995, 2017-09-03:
Thanks.
I just pushed a commit that solves the issue. Probably when I tried to re-install the module for testing, the `start-server.sh` never got deleted and hence, all of my examples still worked!

777arc, 2017-09-03:
Is node.js required to use these blocks?  I can start the flowgraph, but when I go to http://localhost:5006/bokehgui I get the error (in my console):  RuntimeError: node.js is needed to allow compilation of custom models

kartikp1995, 2017-09-03:
Yes. That is because the Waterfall plot in this module is a "custom models". Thanks for reminding. I will add it in CMake and README.

777arc, 2017-09-03:
I just updated the previous comment with my error after installing node.js.

kartikp1995, 2017-09-05:
@777arc Please use NodeJS v6.6 or above. The default Ubuntu package manager installs v4.8. 

Updated the details on README. Thank you.

MattEttus, 2018-01-04:
It looks like start-server.sh is still missing.  Are you sure that commit got pushed?

kartikp1995, 2018-01-20:
@MattEttus I think it's fine without that. I changed the function itself at that time. 

Sorry for delayed response. There's lots of thing going on. 

Let me know if this is fine. For now, I am closing this issue.

Thanks.

kartikp1995, 2018-01-20:
@MattEttus It's fine without that. I changed the function itself at that time.

Sorry for delayed response. There's lots of things going on.

Let me know if this is fine. For now, I am closing this issue.

Thanks.

closed 2018-01-20 by kartikp1995