gnuradio/gr-bokehgui #7: installing gr-bokehgui from pybombs fails

peter-langer, 2017-10-27:
when trying to install gr-bokehgui by pybombs i encountered two problems, both were easily hand-fixed locally:

- in gr-recipes within your gr-bokehgui.lwr recipe the repository is given as [https://github.com/gnuradio/gr-bokehgui](url) but that does not exist => I cloned directly from here, then things continued
- the Gui does not start with the bokeh version installed by default by the recipe => explicitly replacing them by the bokeh version 0.12.6 made things work

kartikp1995, 2017-10-27:
Thanks for the reminder. I will add the recent latest version of Bokeh support by this weekend. Since Bokeh is almost on the verge of getting first version release v1.0, I guess I should stay updated till then.

I did not realize this Pybombs issue. I will drop a PR as soon as possible. Thanks for the update.

Notou, 2020-05-06:
This seems to have been solved so I'm closing the issue.

closed 2020-05-06 by Notou