gnuradio/gr-bokehgui #11: Fails to compile on Mac OS X, can't find a numpy header file

MattEttus, 2018-01-04:
When building on Mac OS X, the build system fails to find a numpy header:

[ 73%] Building CXX object swig/CMakeFiles/_bokehgui_swig.dir/bokehgui_swigPYTHON_wrap.cxx.o
/Users/mettus/gr-bokehgui/build/swig/bokehgui_swigPYTHON_wrap.cxx:4421:10: fatal error: 
      'numpy/arrayobject.h' file not found
#include <numpy/arrayobject.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [swig/CMakeFiles/_bokehgui_swig.dir/bokehgui_swigPYTHON_wrap.cxx.o] Error 1
make[1]: *** [swig/CMakeFiles/_bokehgui_swig.dir/all] Error 2
make: *** [all] Error 2

The actual location is /usr/local/lib/python2.7/site-packages/numpy/core/include but cmake isn't adding that to the compiler search path for some reason.

brjwebb, 2018-01-18:
This answers issue 11 and updates issue 7

This may be more than what you asked for originally, but I thought it may be good for you to be aware of everything I experienced after the initial answer of compilation errors

Methods taken to compile on Linux 16.04:

Installed via clone instructions in tutorial http://kartikpatel.in/GSoC2017/tutorial/

Placed repo in the “include” directory of my pybombs prefix (not sure if that’s a good place, but it’s worth mentioning in case it matters)

mkdir build
cd build

-Start of custom fixes- 

Had to change CMakeLists.txt locally by adding the following to the “include directories” section:

${GNURADIO_ALL_INCLUDE_DIRS}/gnuradio/swig
/usr/local/lib/python2.7/dist-packages/numpy/core/include

Note !: Despite “arrayobject.h” being one directory deeper, the compiler looks for numpy/arrayobject.h so be sure not to go too far in the path name.

cmake ../

make 
sudo make install
sudo ldconfig

At this point gr-bokehgui should be able to completely make and make install, below are instructions to help it function in gnuradio.

Updated gnuradio to latest version in github :

cd mypybombsprefix/src/gnuradio/
git remote update
git pull
cd build
make install

also try ./pybombs update (found in /usr/local/bin ) if possible because it is more thorough in how it updates.

Either create ~/.gnuradio/config.conf or sudo vim ~/prefix/etc/gnuradio/conf.d/grc.conf

Insert : 
[grc]
local_blocks_path=/where/you/put/gr-bokehgui/xml/files/

Install bokeh at the 0.12.6 version (gr-bokehgui is currently using deprecated numpy methods, thus why it’s necessary for this specific version) 

Via pip (conda instructions are both in the tutorial and bokeh page) : 

pip install bokeh==0.12.6

Install node, bokeh-node.js

sudo apt install node
npm install bokehjs 

Note ! : If testing functionality using tutorial or test flowgraphs given, gnuradio-companion may throw an error about the server connection. I found this is fixed by moving one of the blocks, pressing the generate button, and then pressing the green play button.

To double check if server process actually got killed after ending flowgraph in gnuradio :

sudo lsof -i:Port#
sudo kill -9 PID

kartikp1995, 2018-01-20:
@MattEttus This is a typical numpy issue. Can you check 
```
>>> import numpy as np
>>> np.get_include()  
```

One workaround to this is to add a link to the global include directory which points to the real numpy include directory (output of this snippet).

For example: you can check: https://github.com/andersbll/cudarray/issues/25#issuecomment-172257089