gnuradio/gr-bokehgui #13: Widget value does not update

brageee, 2018-01-10:
It seems that for instance GUI Label does not update its field when changed during runtime. I have a variable from one my blocks that I pull with function probe and I would like to display it in the browser, for instance with GUI Label.

kartikp1995, 2018-01-20:
Hello @brageee,
Can you share the simplest flowgraph with the issue. I guess I understand what you are trying to say, but I am unable to reproduce the issue.

brageee, 2018-01-22:
Hi,

The easiest way is simply to include a bokeh gui widget label in the tutorial flowgraph you have included in gr-bokeh, and let the frequency value get displayed also in the label widget. This failed in my flowgraph, i.e. the label value did not update. When I instead used a textbox widget it works, i.e. the textbox updates as the frequency slider changes. 

I also have some "instability" when it comes to the textbox and whether it updates its value in the flowgraph. I tried to control the frequency of a usrp with the textbox widget and it seems that sometimes it works, other times not. Do you have any idea what the issue can be?

Another minor issue I have noticed is that the inital radio button setting is not highlighted. E.g. I have a binary radio button widget, integer 0 or 1 with 0 as inital setting. When the flowgraph starts, neither button is highlighted.

Would it also be possible to implement a widget similar to the QT chooser widget? 

Overall, great work!

Notou, 2020-05-06:
The labels now properly update in the master and maint-3.8 branches.

closed 2020-05-06 by Notou