gnuradio/gr-bokehgui #20: Will not build on master due to latest changes affecting OOTs

mormj, 2020-05-05:
```
/share/gnuradio/grdev/src/gr-bokehgui/lib/time_sink_c_proc_impl.cc: In static member function ‘static gr::bokehgui::time_sink_c_proc::sptr gr::bokehgui::time_sink_c_proc::make(int, double, const string&, int)’:
/share/gnuradio/grdev/src/gr-bokehgui/lib/time_sink_c_proc_impl.cc:38:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::bokehgui::time_sink_c_proc_impl]()’ from ‘std::shared_ptr<gr::bokehgui::time_sink_c_proc_impl>’ to ‘gr::bokehgui::time_sink_c_proc::sptr’ {aka ‘boost::shared_ptr<gr::bokehgui::time_sink_c_proc>’}
   37 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   38 |         (new time_sink_c_proc_impl(size, samp_rate, name, nconnections));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::bokehgui::time_sink_c_proc_impl>
```
Changes to get OOT compatible with 3.9 (current master) should be documented here:
https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

Notou, 2020-05-06:
Yes, the change in shared pointers broke compatibility. Thanks for the link.

Notou, 2020-05-07:
It should work now.

Notou, 2020-06-19:
This now needs to be updated to support the new pybind system.

Notou, 2021-11-26:
The master branch now supports GR 3.9.4.
The change to pybind11 is done.

closed 2021-11-26 by Notou