gnuradio/gr-bokehgui #28: Operation relies on deprecated event loop

Notou, 2020-05-29:
The updating of the plots uses a bokeh client event loop, in the form of the session.loop_until_closed() function. 
But this function is deprecated in bokeh 1.x and removed in 2.x, so we will need to find a way to stop using it if we want to use up to date dependencies.

Current operation:

1. The bokeh server is started in command line in a external server with an empty document (blank page)
2. The top block initialisation fills up a separate document with all the wanted plots, widgets, layouts and callbacks.
3. A session is created programatically by connecting to the running server and set to use the previously created document.
4. session.loop_until_closed() is used to serve the callbacks (mainly updates to the plots data)

This means that, currently, the proper document is shown for just one named session, that it why, if you connect to a running server and use the wrong id in ?bokeh-session-id=<id> or forget it entirely, you get a blank page without errors.

An event loop is needed to handle callbacks to update the plots and react to user input on the widgets; session.loop_until_closed() was the only way to run one outside of the bokeh server and I don't want to have to run the graph running bokeh serve, so we will have to run the bokeh server inside our top block process.

Proper bokeh operation is actually quite far from what is done right now so we will need substantial changes in the way documents are created and handled.
Documents consist mainly of a layout (a list of gui elements) and a callback list. The documents cannot be directly copied, the layout can only be transferred between documents but cannot be in more than one at once and the callback list does not move.
We need a function that will be able to create a new document when called and be called multiple times (so it cannot be the __init__() of the top block). It will also probably need to create new widget and plot objects every time (not possible to store a list).

quentinmit, 2020-08-20:
Most nontrivial flowgraphs can't run in a multi-doc model. I have looked at "bokeh serve" several times and it doesn't seem to have any way to present the same session to every client, as gr-bokehgui does now.

Right now I'm experimenting with a model where there's one flowgraph instance for the server, and then new copies of the bokeh blocks are created and connected for each client. That's not great, though. State still doesn't sync across clients.

Notou, 2020-08-20:
Thanks for relaunching me on that :)
I've done some experimenting too, and I've come up with a similar model where only the bokeh half of the blocks gets created with a new connection. That way most of the processing gets shared.
The annoying thing is that, as you mentioned, bokeh parameter updates are not propagated to all the connected clients, only the last one to be loaded. It may be an ok ish compromise since, if you are not the last connected client, you can always reload the page to get the up to date version.

I'll try to get my code in order and share it with you, if you want to take a look at it.

quentinmit, 2020-08-29:
> * Ah, yes, good to know. I would prefer not using it though, this feels like asking for more trouble.
Agreed, also there's apparently a big performance penalty for using an out-of-process server.
> * You're right, to get GUI in sync between two tabs, one just need to open the same session twice. This means we don't have to worry too much about state sync between clients. I won't try to implement now one of the automatic methods you described though since we can easily do it manually for now and I want to finish this first. But you're welcome to do it if you want :)
> * Nice. Does this all work with 1.4, or is only for 2.0?
I have not tested my branch with Bokeh 1.4 (which is why I haven't actually opened a pull request...)
> * It looks like what I expected from your description. And apparently there is another bad aspect of this approach: the need to stop the flowgraph to connect the new blocks, leading to delays and sample loss.
I don't think you can create only the bokeh half of the blocks as in your branch; the blocks assume that only one thing is pulling samples from them, because they drop the samples as soon as they have been read. (So I'd expect your waterfalls/spectrograms to get half the frame rate if they're open twice, etc.)

Notou, 2020-09-22:
> I have not tested my branch with Bokeh 1.4 (which is why I haven't actually opened a pull request...)

It would seem that everything works for 1.4 and 2.0, except for the waterfall changes that only work with 2.1 (not 2.0).

> I don't think you can create only the bokeh half of the blocks as in your branch; the blocks assume that only one thing is pulling samples from them, because they drop the samples as soon as they [...]

This is correct (and I did not realize that at the time), but I'm not sure it is actually a problem: 

1. In most use cases, the flow of samples is faster than the display speed, so you miss quite a lot of samples anyway. (except for waterfall)
2. If you really need to see everything, you can still open with the same session, reverting to the previous way of using the module.
3. We should be able to modify the get_plot_data function to not drop samples as soon as they are read, at the same time fixing the issue of buffer delay, when the cpp side buffer fill up quicker than it empties and you get a massive delay (10s of s) between changing something in the graph and seeing its effect on the display.

Notou, 2021-07-06:
I finally got time to work on this and got everything working with the latest version of Bokeh.
So I think I can close this issue, since the base problem is solved.

Notou, 2021-11-26:
To get back on my last comments:

> We should be able to modify the get_plot_data function to not drop samples as soon as they are read, at the same time fixing the issue of buffer delay, when the cpp side buffer fill up quicker than [...]

The delay issue I'm talking about there is not from the cpp side buffer that fills up. The queue there is actually fairly small.
It actually comes from the browser itself. When it is not able to process all the plot updates in real time (if the update period is too high) it starts to queue them up. 
To not see this queuing, one needs to greatly reduce the plots refresh rate, especially with many plots, so you really miss a lot of samples. Hence I really feel it's not an issue.

closed 2022-03-03 by Notou