gnuradio/gr-bokehgui #29: Unable to serve bokeh document

mormj, 2020-06-01:
I see the following behavior on both the maint-3.8 and master branches from running test_bokehgui example

Are there steps that I am missing?
```
josh@josh-clean1910:/share/gnuradio/gr38/src/gr-bokehgui$ /usr/bin/python3 /home/josh/.vscode/extensions/ms-python.python-2020.3.71659/pythonFiles/lib/python/debugpy/wheels/debugpy/launcher /share/gnuradio/gr38/src/gr-bokehgui/examples/test_bokehgui.py 
2020-06-01 10:55:32,217 Starting Bokeh server version 1.4.0 (running on Tornado 5.1.1)
2020-06-01 10:55:32,218 Host wildcard '*' will allow connections originating from multiple (or possibly all) hostnames or IPs. Use non-wildcard values to restrict access explicitly
2020-06-01 10:55:32,218 User authentication hooks NOT provided (default user enabled)
2020-06-01 10:55:32,219 Bokeh app running at: http://localhost:5007/bokehgui
2020-06-01 10:55:32,219 Starting Bokeh server with process id: 28275
2020-06-01 10:55:46,541 WebSocket connection opened
2020-06-01 10:55:46,541 Uncaught exception GET /bokehgui/ws (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:5007', method='GET', uri='/bokehgui/ws', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/websocket.py", line 546, in _run_callback
    result = callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bokeh/server/views/ws.py", line 122, in open
    raise ProtocolError("No bokeh-protocol-version specified")
ProtocolError: No bokeh-protocol-version specified
2020-06-01 10:55:46,542 WebSocket connection closed: code=None, reason=None
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/josh/.vscode/extensions/ms-python.python-2020.3.71659/pythonFiles/lib/python/debugpy/wheels/debugpy/__main__.py", line 45, in <module>
    cli.main()
  File "/home/josh/.vscode/extensions/ms-python.python-2020.3.71659/pythonFiles/lib/python/debugpy/wheels/debugpy/../debugpy/server/cli.py", line 429, in main
    run()
  File "/home/josh/.vscode/extensions/ms-python.python-2020.3.71659/pythonFiles/lib/python/debugpy/wheels/debugpy/../debugpy/server/cli.py", line 266, in run_file
    runpy.run_path(options.target, run_name=compat.force_str("__main__"))
  File "/usr/lib/python3.7/runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)
  File "/usr/lib/python3.7/runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/share/gnuradio/gr38/src/gr-bokehgui/examples/test_bokehgui.py", line 251, in <module>
    main()
  File "/share/gnuradio/gr38/src/gr-bokehgui/examples/test_bokehgui.py", line 232, in main
    url = "http://localhost:" + port + "/bokehgui")
  File "/usr/local/lib/python3.7/dist-packages/bokeh/client/session.py", line 163, in push_session
    session.push(document)
  File "/usr/local/lib/python3.7/dist-packages/bokeh/client/session.py", line 386, in push
    raise IOError("Cannot push session document because we failed to connect to the server (to start the server, try the 'bokeh serve' command)")
OSError: Cannot push session document because we failed to connect to the server (to start the server, try the 'bokeh serve' command)
```

Notou, 2020-06-04:
This is strange.
I use the exact same bokeh and tornado versions and I've never seen this issue.

It seems you are using python 3 to run the graph but the tornado and bokeh error messages point to a python2.7 directory. That may be one of the issues.

fikrifa0, 2021-01-30:
I have tried from generated python file from examples/tutorial.grc, by `python3 -m bokeh serve tutorial.py` using bokeh 1.4.0 in virtualenv,

the bokeh server started but it returns
> blank page
> with page title Bokeh Application,
> and success response from console in inspect element.

Notou, 2021-01-30:
You don't need to use bokeh serve to launch the file, a direct `python3 -m tutorial.py` should work.
All the bokeh handling is done inside the python script.

fikrifa0, 2021-02-02:
[image]

Thanks, 3.8-bokeh-2 worked!

closed 2021-11-26 by Notou