gnuradio/gr-bokehgui #32: tutorial.grc displayed unsuccessfully on the web page ("SyntaxError: Unexpected identifier")

Csysongyi, 2020-10-18:
When I run the tutorial.grc file in example, it can't run successfully and it shows : "500: Internal Server Error" on the web page

Here is log : 

RuntimeError: /home/csygnu/.local/lib/python3.6/site-packages/bokeh/server/static/js/compiler.js:271423
        async remove(item) {
              ^^^^^^
SyntaxError: Unexpected identifier
    at Object.exports.runInThisContext (vm.js:76:16)
    at Module._compile (module.js:542:28)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3

2020-10-18 16:11:29,555 500 GET /bokehgui?bokeh-session-id=tutorial (127.0.0.1) 633.53ms
2020-10-18 16:11:29,660 404 GET /favicon.ico (127.0.0.1) 0.37ms
Exiting the simulation. Stopping Bokeh Server

>>> Done

Notou, 2020-10-23:
Hi, sorry about the delay,
Could you please tell me what version of bokeh, gnuradio and bokehgui you are using?

Notou, 2021-11-26:
I'm closing this since I've not had any feedback for a year.

closed 2021-11-26 by Notou