gnuradio/gr-bokehgui #33: gnuradio 3.9 requirements should be mentioned in the readme file

planewave, 2020-11-11:
since the v3.8 support has been dropped.

Notou, 2020-11-11:
The v3.8 support has not been dropped. It does not even work with 3.9.

What made you think that?

planewave, 2020-11-11:
found the commit comment below
[image]
in `gr-bokehgui/CMakeLists.txt` line 90
```
find_package(Gnuradio "3.9" REQUIRED runtime fft)
```

Notou, 2020-11-11:
You are right, my mistake. 
The readme should mention to use the branch maint-3.8, and not master.

planewave, 2020-11-11:
OK, here is what I did. (Ubuntu 20.04)

1. install maint-3.8 using pybombs with `gnuradio-default` recipe 
2. change `Gnuradio "3.9"` in the CMakeLists.txt line 90 to `Gnuradio "3.8"` . Otherwise it still looking for 3.9, like
```
CMake Error at CMakeLists.txt:90 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.9".

  The following configuration files were considered but not accepted:

    /home/user/gnuradio/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.2.0
```
3. then it was looking for LOG4CPP
```
-- Looked for LOG4CPP libraries named log4cpp.
CMake Error at /home/skycope/gnuradio/lib/cmake/gnuradio/FindLOG4CPP.cmake:46 (message):
  Could NOT find LOG4CPP library
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /home/skycope/gnuradio/lib/cmake/gnuradio/GnuradioConfig.cmake:25 (find_dependency)
  CMakeLists.txt:90 (find_package)
```
so I installed it by
```
sudo apt install liblog4cpp5-dev
```
cmake passed
4. then make it but got something like https://github.com/gnuradio/gr-bokehgui/issues/20
```
gr-bokehgui/lib/vec_sink_f_proc_impl.cc:36:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::bokehgui::vec_sink_f_proc_impl]()’ from ‘boost::shared_ptr<gr::bokehgui::vec_sink_f_proc_impl>’ to ‘gr::bokehgui::vec_sink_f_proc::sptr’ {aka ‘std::shared_ptr<gr::bokehgui::vec_sink_f_proc>’}
   35 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   36 |         (new vec_sink_f_proc_impl(vlen, name, nconnections));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         boost::shared_ptr<gr::bokehgui::vec_sink_f_proc_impl>
make[2]: *** [lib/CMakeFiles/gnuradio-bokehgui.dir/build.make:63: lib/CMakeFiles/gnuradio-bokehgui.dir/vec_sink_f_proc_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:274: lib/CMakeFiles/gnuradio-bokehgui.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

stuck again

Notou, 2020-11-12:
It looks like you are still using the master branch from bokehgui.
You have to use maint-3.8 for both gnuradio and gr-bokehgui. Sorry this wasn't clearer.

planewave, 2020-11-12:
thank you for the clarification. after installation of `liborc-dev` and `python-numpy`, it passed the make.
but failed int the make test
```
0% tests passed, 6 tests failed out of 6

Total Test time (real) =   0.74 sec

The following tests FAILED:
	  1 - qa_time_sink_f (Failed)
	  2 - qa_time_sink_c (Failed)
	  3 - qa_freq_sink_f (Failed)
	  4 - qa_freq_sink_c (Failed)
	  5 - qa_waterfall_sink_f (Failed)
	  6 - qa_waterfall_sink_c (Failed)
Errors while running CTest
make: *** [Makefile:130: test] Error 8
```
any suggestions?

Notou, 2020-11-12:
Nothing comes to mind. Do you have the log of the failure?

And also, you should be able to skip the `make test` step. Does it work with grc and an example graph? Like test_bokehgui.grc.

planewave, 2020-11-12:
So I checked the log
```
  File "/home/skycope/dev/gr-bokehgui/python/qa_time_sink_f.py", line 24, in <module>
    from bokehgui_swig import time_sink_f_proc
  File "/home/skycope/dev/gr-bokehgui/build/swig/bokehgui_swig.py", line 15, in <module>
    import _bokehgui_swig
ImportError: libvolk.so.2.0: cannot open shared object file: No such file or directory
```
but I have `libvolk2-dev` installed already

then I skipped it, `sudo make install` nothing complained.
tested with `test_bokehgui.grc`, said **No module named 'bokehgui'**
[image]

Notou, 2020-11-15:
Hum, swig is making waves ... And that's had to debug. :/
Do you have a clean install of everything?

If you do and it still does not work, you may be able to see more debug info of what is actually causing swig to fail: in the installed python files related to swig, the thing that imports the bokehgui_swig is inside of a try catch statement that hides the actual error message. If you remove that, and execute it, you could get a more useful exception.

Notou, 2021-11-26:
I'm closing this since we've not had any updates in a year.

I hope it worked. If not, there is the possibility to go to master and GR 3.9, since I managed to get switch to pybind11.

closed 2021-11-26 by Notou