gnuradio/gr-bokehgui #34: GrSwig missing in during cmake

planewave, 2020-11-11:
I am new to the cmake thing, so this may be a silly question. when I run the cmake, it shows the error below 
```
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
```

after googling it, it appears that the `cmake/Modules/GrSwig.cmake` is missing. but what can I do?
thank you

Notou, 2020-11-11:
Could you tell me what version of gr and what branch of bokehgui you are using?

planewave, 2020-11-11:
both are in the master branches 
gr3.9 as mentioned in the previous post

Notou, 2020-11-11:
OK, the thing is, 3.9 dropped the use of swig in favor of pybind11, and I've not yet figured how to properly port bokehgui to the new system.
This means that master is broken at the moment. This needs to be reflected in the readme.

Please use maint-3.8 for both gnuradio and bokehgui.

closed 2020-11-12 by planewave

GlenLangston, 2020-12-23:
Crap.  I can't proceed without 3.9 and swig.

Please disable 3.9.

GlenLangston, 2020-12-23:
SDRPLay does not work in 3.8
I need SDRPlay support.

What a mess.

marcoferrara10, 2021-01-03:
Hello everyone,
did you solve this issue? Because I've the amazing idea to compile everything manually XD...thank you

Notou, 2021-01-03:
Hi, I have not had the time to finish porting to 3.9 unfortunately. I hope to be able to dedicate some time to do that when I finish my thesis in the next months.
But 3.8 should work fine.

marcoferrara10, 2021-01-03:
Dear @Notou thank you so muck for your answer! I've understood now the issue and I'm going back to the 3.8! 
Thank you again!

Notou, 2021-01-03:
My pleasure, it's nice to see people using this module.

mjihad-75, 2021-01-28:
Hello everyone,
I'm working with 3.8 and I'm getting the same error, do you solve this issue? 
Thank you!!

Notou, 2021-01-28:
Hi @mjihad-75, are you using the main-3.8 branches for both gnuradio and this module?

ahmedshahein, 2021-01-30:
Hello,
Any update regarding the same issue? I am using Ubuntu 18.04.

Notou, 2021-01-30:
Hello,
If you have a swig problem using both the maint-3.8 branches of gnuradio and the gr-bokehgui, please put it in a new issue with more details because this one was related to not using a working combination of repo branches.

jburnett007, 2021-03-06:
Hello, I had this GrSwig error today.  Running Ubuntu 18.04. Tried to upgrade to GNU Radio 3.8 but it just went to 3.9. It would be great if this Gr Swig error could be please fixed. I am stuck..  I am trying hard to gr-kiwisdr going. 

GrSwig

CMake Error at swig/CMakeLists.txt:40 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

Many thanks.

Notou, 2021-03-06:
Hello, this module needs quite a bit of work to port it to 3.9 and I won't have time to do it myself in the next couple of months.
It works with 3.7 and 3.8 though.
What's the link with gr-kiwisdr?

jburnett007, 2021-03-08:
Thank you. I did uninstalled Version 3.9 and installed GnuRadio 3.8 on my Ubuntu 18.04 and it did solve this swig problem. I am trying to get gr-kiwisdr working. 
https://github.com/hcab14/gr-kiwisdr
So now cmake ../ works   (-:
But getting a error now when I run the next step   make install   .. But that is something to do with gr-kiwsdr
Hope all your studies go well..

jburnett007, 2021-04-29:
Hello Notou,
In the mean time I had to upgrade by Ubuntu to 20.04 and am using GNURadio 3.9 
But still have been unable to get gr-kiwisdr to compile. )-: 
Hope you can find the time to work on this. 
Thanks.

Ahmedmba37, 2022-08-20:
Dears , 
as new guy for SDR , I have got below errors while installing the Gun-GSM , 
I ran this command "sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default"
than the installation broken and i have got below errors lines :
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find load file:
    GrSwig
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
can any one help ?

Notou, 2022-08-21:
Hi, 
The command you are trying to run is for installing gnuradio, not gr-bokehgui, which is an out of tree module.
So please raise the issue in the gnuradio repository.
If I where to guess, you probably need to install the swig package.

greyhat94, 2022-09-23:
I'm using gnuradio 3.10.3.0 and I've installed gr-gsm, but I keep getting the error concerning the swig file. I've seen that many people are experiencing the same issue. Is possible that is there no solution to fix it?

luciapintor, 2022-10-05:
Hi, I don't know if it is still useful @greyhat94, but it might be a version incompatibility. You can try to switch branch in the repo (git checkout maint-3.10). By default the clone repo is on branch 3.8 and your gnuradio is 3.10.

Notou, 2022-10-05:
Hi, it is indeed certainly a version mismatch issue, with gr-gsm expecting gnuradio 3.8, that uses swig, and gnuradio 3.10 that you have install, that uses pybind11.

But please raise that issue with the gr-gsm team.
This here is the repository for gr-bokehgui, that has nothing related to gr-gsm.

greyhat94, 2022-10-05:
The right command I have to run is this: get clone maint-3.10
https://link.it/, right?