gnuradio/gr-bokehgui #35: Assertion failed

fikrifa0, 2021-01-28:
I got this when inserting bokeeh gui sinks block on gnuradio

Assertion "len(placement) == 4 or len(placement) == 2" failed.

Running with gr-bokehgui branch maint 3.8 and gnuradio 3.8 from ppa (gnuradio 3.8.2.0)

Thanks for any help!

Notou, 2021-01-28:
This comes from the configuration of the widget `placement` parameter.

It needs to be a tuple of either length 2 or 4, like (1,0) or (2,0,1,2), similar to how you would configure QT widgets with the `GUI Hint` parameter.

This is described in more detail in the last section of the [tutorial](https://kartikpatel.in/GSoC2017/tutorial/)

fikrifa0, 2021-01-29:
Thanks for this, now I finding solution about top_block_sptr has no attribute doc after configuring python path

> This comes from the configuration of the widget `placement` parameter.
>
> It needs to be a tuple of either length 2 or 4, like (1,0) or (2,0,1,2), similar to how you would configure QT widgets with [...]

closed 2021-01-29 by fikrifa0

fikrifa0, 2021-01-29:
In my case, Bokeh GUI generate options appear by copying installed library to python default library path
`sudo cp /usr/local/lib/python3/dist-packages/bokehgui /usr/lib/python3/dist-packages -r `

Notou, 2021-01-29:
You mean that if the lib is in the PYTHONPATH but is somewhere else than the default library, it's not detected by gnuradio?

fikrifa0, 2021-01-30:
Yes, I have tried run this command on terminal before,
`export PYTHONPATH=/usr/local/lib/python3/dist-packages`
and adding some sys append to .pythonrc, but the Generate Options still four blocks.
Then, I try to copy the library to the default path that looks similar, and it appears.

Notou, 2021-01-30:
Ok, that is strange. 
I don't remember checking this case so there could be an issue in the way gnuradio checks for this module.

But looking at the place gnuradio does the [check](https://github.com/gnuradio/gnuradio/blob/74ae2184b1f5b025418bf57ba604f3e7a1f97523/grc/core/utils/hide_bokeh_gui_options_if_not_installed.py), it's a simple python `import`, same as what's used by OOT blocks. So it could also be that the `export` did not add the proper path to the PYTHONPATH seen by grc.  
I usually do the export to add to the PYTHONPATH in my .bashrc file and launch grc in a fresh terminal (it sometimes does not work when starting if from a shortcut).