gnuradio/gr-bokehgui #36: Frequency sink plot can't show in mobile web client

fikrifa0, 2021-03-02:
I run my simple bokehgui frequency sink without widget. It appears on desktop web client but failed to appear on mobile web client. I have added `allow_websocket_origin=['*']` to `Server` in `bokehgui/utils.py`.

I have tried to modify `bokehgui/utils.py` by added an example from bokeh repository in `make_doc(doc)` function, and it can appears on mobile web cilent.

So what can I do to make bokehgui frequency sink appears on mobile web client?

Thanks in advance.

Notou, 2021-03-03:
I've never tried to run bokehgui on a mobile so it's new territory.

Could you link the example you used successfully?

ffrmns, 2021-03-05:
I use the code from examples/howto/server_embed/standalone_embed.py
standalone_embed.py

Notou, 2021-03-05:
It's indeed strange that you could see the temperature plot, and not the plots inside `plot_list`.
Especially since the frequency sink is the same type of bokeh plot.

Could the duplicate `doc.add_root` cause an issue there? Maybe only the first one is taken into account?

Maybe it's the type of layout used by bokehgui that the mobile browser doesn't like.
Do the other types of bokehgui plots work on mobile?

Sorry I'm not able to test things on my side at the moment.

Notou, 2021-03-08:
I've repaired my bokehgui installation and I quickly tried the test_bokehgui example on mobile. 
(On the main-3.8 branch, not the one supporting bokeh 2.0).
And the default example works on my phone, it's glitchy and sometimes fails but it works.
But including sinks other than the time plot makes my phone unable to connect to the server, like you describe. (the widgets don't seem to cause problems)

It's very mysterious, I'm not sure I can help you debug this, but at least I could reproduce your issue.

ffrmns, 2021-04-13:
Trying with another phone,
[image]

It's work with another phone.

closed 2021-10-30 by ffrmns