gnuradio/gr-bokehgui #37: possible dependency issues between bokeh 1.4->nodejs->python3?

muaddib1984, 2021-03-07:
Hi,
Could you please provide guidance on the dependency versions for bokeh to support GNURadio maint-3.8?
Since the project has been pinned at bokeh v1.4 for a while (which seems old) i think it would be nice to have some documentation that supports the install process for bokeh and it's dependencies, especially on a newer build like Ubuntu18.04LTS with GNURadio 3.8. 
I'd be happy to write up the install docs for you once i get it working.
I've had this working in the past on GNURadio 3.7.14.0 with python 2.7 and i remember quite a few issues finding the right nodejs version.
so far I've installed the following:
gr-bokehgui from source (git branch maint-3.8)
bokeh 1.4 (via pip3)
nodejs - i've tried the current Ubuntu apt repo version which is 8.x, removed that, and now i'm on v6.10.1 per STDOUT errors when no version of nodejs is installed. This should match the bokeh 1.4 dependencies.

The server starts when i run the examples, however the issue i'm facing is when i conect to the server from my web browser i get this error:
Executing: /usr/bin/python3.6 -u /home/muaddib/prefix/gr38/src/gr-bokehgui/examples/test_bokehgui.py

2021-03-07 11:59:55,213 Starting Bokeh server version 1.4.0 (running on Tornado 6.1)
2021-03-07 11:59:55,213 Host wildcard '*' will allow connections originating from multiple (or possibly all) hostnames or IPs. Use non-wildcard values to restrict access explicitly
2021-03-07 11:59:55,214 User authentication hooks NOT provided (default user enabled)
2021-03-07 11:59:55,216 Bokeh app running at: http://localhost:5006/bokehgui
2021-03-07 11:59:55,216 Starting Bokeh server with process id: 10335
2021-03-07 11:59:55,960 WebSocket connection opened
2021-03-07 11:59:55,962 ServerConnection created
BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
BokehDeprecationWarning: ClientSession.loop_until_closed is deprecated, and will be removed in an eventual 2.0 release. Run Bokeh applications directly on a Bokeh server instead. See:

    https//docs.bokeh.org/en/latest/docs/user_guide/server.html

2021-03-07 12:00:00,307 Uncaught exception GET /bokehgui (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:5006', method='GET', uri='/bokehgui', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/home/muaddib/.local/lib/python3.6/site-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/home/muaddib/.local/lib/python3.6/site-packages/tornado/gen.py", line 775, in run
    yielded = self.gen.send(value)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/server/views/doc_handler.py", line 61, in get
    template_variables=session.document.template_variables)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/embed/server.py", line 227, in server_html_page_for_session
    bundle = bundle_for_objs_and_resources(None, resources)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/embed/bundle.py", line 184, in bundle_for_objs_and_resources
    ext = bundle_models(models)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 338, in bundle_models
    _bundle_cache[key] = bundle = _bundle_models(custom_models)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 551, in _bundle_models
    custom_impls = _compile_models(custom_models)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 528, in _compile_models
    compiled = nodejs_compile(impl.code, lang=impl.lang, file=impl.file)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 111, in nodejs_compile
    output = _run_nodejs([compilejs_script], dict(code=code, lang=lang, file=file, bokehjs_dir=bokehjs_dir))
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 475, in _run_nodejs
    return _run(_nodejs_path(), argv, input)
  File "/home/muaddib/.local/lib/python3.6/site-packages/bokeh/util/compiler.py", line 470, in _run
    raise RuntimeError(errout.decode('utf-8'))
RuntimeError: /home/muaddib/.local/lib/python3.6/site-packages/bokeh/server/static/js/compiler.js:271423
        async remove(item) {
              ^^^^^^
SyntaxError: Unexpected identifier
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:542:28)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:393:7)
    at startup (bootstrap_node.js:150:9)

2021-03-07 12:00:00,309 500 GET /bokehgui (127.0.0.1) 480.29ms

Any help would be much appreciated!

Notou, 2021-03-08:
Hi,
Bokeh 1.4 is indeed a bit old. There was some work needed to upgrade the update loop in bokehgui to adapt to dropped API in Bokeh 2.0 (the `BokehDeprecationWarning: ClientSession.loop_until_closed is deprecated, and will be removed in an eventual 2.0 release.` bit).
This is part of the 3.8-bokeh-2 but it also necessitated changes on the gnuradio side. These were pulled recently and should be included in the next 3.8.3 release (I don't know when though).

As for node, I have it working with v10.19.0 at the moment, but it should also work with earlier versions.

muaddib1984, 2021-04-30:
Sorry for the delay on this. So am I understanding correctly that there is a combination of certain versions of bokeh and node that will currently work with GNURadio maint-3.8 (with gr-bokehgui functions) or are we still waiting for the pulled code to get merged?
Thanks!

<end transmission>

Notou, 2021-05-03:
Since the pulled code got merged, there are currently two combinations available:
- One with gr 3.8.2, the maint-3.8 branch bokehgui, and bokeh <2.X
- And the latest gr 3.8.3 (or the maint-3.8 branch), the 3.8-bokeh-2 branch bokehgui, and bokeh > 2.X

And for both, you need nodejs, most versions should work.

I should have some time "soon" to get the 3.8-bokeh-2 branch into the maint-3.8 branch, with update to the readme to clear the install instruction, and start on working with 3.9.

muaddib1984, 2021-05-03:
Thank you so much!
I’m hoping to get a chance to install this on a fresh system later this week. Once I get it working If i find anything noteworthy in the dependency/build process I’ll report back. 
Thanks again
<end transmission>

Notou, 2021-11-26:
Closing this, but there is now also the following combination:

gr 3.9 (tested with 3.9.4, should also work with master), the master branch bokehgui, and bokeh <2.X

closed 2021-11-26 by Notou