gnuradio/gr-bokehgui #38: AttributeError: module 'bokehgui' has no attribute 'BokehLayout'

Mayflyworld, 2021-05-13:
I see the following behavior on both the maint-3.8 and master branches from running bokehgui example

dragon@dragon:~/Desktop/myoot/gr-bokehgui/examples$ python3 tutorial.py
2021-05-12 22:31:19,153 Starting Bokeh server version 1.4.0 (running on Tornado 4.5.1)
2021-05-12 22:31:19,154 Host wildcard '*' will allow connections originating from multiple (or possibly all) hostnames or IPs. Use non-wildcard values to restrict access explicitly
2021-05-12 22:31:19,154 User authentication hooks NOT provided (default user enabled)
2021-05-12 22:31:19,155 Bokeh app running at: http://localhost:5006/bokehgui
2021-05-12 22:31:19,156 Starting Bokeh server with process id: 4484
2021-05-12 22:31:19,857 WebSocket connection opened
2021-05-12 22:31:19,858 ServerConnection created
BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
Traceback (most recent call last):
  File "tutorial.py", line 241, in <module>
    main()
  File "tutorial.py", line 226, in main
    tb = top_block_cls(doc)
  File "tutorial.py", line 157, in __init__
    input_t = bokehgui.BokehLayout.widgetbox(self.widget_lst)
AttributeError: module 'bokehgui' has no attribute 'BokehLayout'
swig/python detected a memory leak of type 'gr::fft::window::win_type *', no destructor found.

Notou, 2021-05-20:
What is your gnuradio version?

MayflyWorldfl, 2021-05-21:
> 您的gnuradio版本是什么？

Thank you for your reply, my gnuradio companion 3.8.1.

Notou, 2021-05-21:
Hum, this error should have been fixed in 3.8.1.
I just checked, the relevant [commit](https://github.com/gnuradio/gnuradio/pull/3316) was indeed pulled for this version.

Please make sure that you truly have the 3.8.1 release, and not one of the release candidates.
Alternatively, you could also use the 3.8.2 release.

MayflyWorldfl, 2021-05-23:
Thank you very much, I will try again!

MayflyWorldfl, 2021-05-23:
> Hum, this error should have been fixed in 3.8.1.
> I just checked, the relevant [commit](https://github.com/gnuradio/gnuradio/pull/3316) was indeed pulled for this version.
>
> Please make sure that you [...]

Thank you very much, I will try again!