gnuradio/gr-bokehgui #42: Attribute Error

pablopfff, 2021-11-02:
Hi, I installed the gr-bokehgui from the source code in the branch maint-3.8 and followed all the instructions, but when I run the flowgraph I have this error in the console
Traceback (most recent call last):
  File "/home/gnuradio/gr-bokehgui/examples/tutorial.py", line 245, in <module>
    main()
  File "/home/gnuradio/gr-bokehgui/examples/tutorial.py", line 213, in main
    serverProc, port = bokehgui.utils.create_server()
AttributeError: module 'bokehgui.utils' has no attribute 'create_server'

Also, the make test fails at the fourth test, the qa_freq_sink_c test with error 8

Notou, 2021-11-02:
Hi, 
To be clear, you installed the branch maint-3.8, and you are using the latest 3.8 gnuradio version? > 3.8.3

pablopfff, 2021-11-02:
> Hi, To be clear, you installed the branch maint-3.8, and you are using the latest 3.8 gnuradio version? > 3.8.3

Im using gnuradio 3.8.3.1

Notou, 2021-11-02:
It seems you are using the right versions, so this error should not happen: 
'bokehgui.utils' has indeed no attribute 'create_server'. but grc in 3.8.1 does not generate top_block files that call this function.

That may have been remains from a file generated with an old version of grc. If that happens again, try removing the python file and regenerate your flowgraph.

You do indeed need nodejs to run flowgraphs (I believe it's only if you use watefall sinks). It seems the readme is lacking in that regard, sorry.

Don't worry about the make test outcome.

pablopfff, 2021-11-02:
Thanks for the help. Now it seems to work with the tutorial.

closed 2021-11-02 by Notou