gnuradio/gr-bokehgui #44: ss is replacing netstat

kb3gtn, 2021-12-23:
On newer linux distro's the netstat command is being replaced with the ss command.   This breaks the check-port.sh script currently used.

Recommend doing something like this to support both netstat and ss commands for bokeh port discovery.

<pre>
# Use netstat if available, else try ss command
if ! command -v netstat &> /dev/null
then
	ss -netul | awk '{ print $5 }' | awk -F':' '{ print $NF }' | tail -n +3 > temp_netstat
else
	netstat -netul | awk '{ print $4 }' | awk -F':' '{ print $NF }' | tail -n +3 > temp_netstat
fi
</pre>

Notou, 2022-01-11:
Oh, thank you for this info!
We had stumbled upon this issue but didn't find what the source was.
I'll try to use your code to update the script (You can also make a PR, if you have the time :) )

Notou, 2022-01-27:
It should be fixed in master now, to use with gr 3.9.5 and 3.10.

Thanks for the snippet!

closed 2022-02-09 by Notou