gnuradio/gr-bokehgui #45: AttributeError: 'gnuradio.gr.gr_python.top_block_pb' object has no attribute 'doc'

tkircher, 2022-01-26:


Notou, 2022-01-26:
Ah, yes, it appears I had not updated the master branch with the latest changes of gr 3.9.5 and 3.10.
Upgrades to the GRC top_block generation were made to allow for up-to-date bokeh runtime loop.

I've a branch already ready for that in my own fork, and have made a [PR](https://github.com/gnuradio/gr-bokehgui/pull/46) to get it into master.
I'll check that everything still works fine (It should) and pull it soon (this evening hopefully).
If you are in a hurry, feel free to test it on your side.

tkircher, 2022-01-26:
This fixed it, thanks. I ended up making this modification to allow all access once again, for anyone else who might be reading this thread:

in bokehgui/utils.py:

```
-    server = Server(app)
+    server = Server(app, port=5006, allow_websocket_origin=["*"])
````

closed 2022-01-26 by tkircher

Notou, 2022-01-27:
The PR is merged now, so you can get back to using the main repository.

wanglifu96, 2022-05-20:
> GNU Radio 版本 3.9.5.0，Python 3.8.10。来自 main 的 gr-bokehgui。一切似乎都构建得很好，但没有一个 bokehgui 模块将运行。我试图在 hier_block2.py 中将其短路，但随后出现关于未调用`__init__`派生类的错误。我将尝试降级到 GR 3.8，但它太老了，我已经从 GR 3.10 降级以尝试找到兼容性。

Hello, have you solved this problem? I'm confused
[image]