gnuradio/gr-bokehgui #58: AttributeError: 'gnuradio.gr.gr_python.top_block_pb' object has no attribute 'plot_lst'

AllisonOge, 2022-10-12:
Hello,

I get the AttributeError with bokehgui when I run the following flowgraph in gnuradio-3.10 in a docker container from the [`m1mbert/cxlb-gnuradio-3.10`](https://hub.docker.com/r/m1mbert/cxlb-gnuradio-3.10).

[image]

The complete console output is 
``` shell
<<< Welcome to GNU Radio Companion 3.10.3.0 >>>

Block paths:
	/cortexlab/toolchains/current/share/gnuradio/grc/blocks

Generating: '/root/testing_bokehgui.py'

Executing: /usr/bin/python3 -u /root/testing_bokehgui.py

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "/root/testing_bokehgui.py", line 207, in <module>
    main()
  File "/root/testing_bokehgui.py", line 185, in main
    tb = top_block_cls()
  File "/root/testing_bokehgui.py", line 90, in __init__
    self.bokehgui_waterfall_sink_x_0_plot = bokehgui.waterfall_sink_c(self.plot_lst, self.bokehgui_waterfall_sink_x_0, update_time = 100,
  File "/cortexlab/toolchains/current/lib/python3.9/site-packages/gnuradio/gr/hier_block2.py", line 88, in __getattr__
    return getattr(self._impl, name)
AttributeError: 'gnuradio.gr.gr_python.top_block_pb' object has no attribute 'plot_lst'

>>> Done (return code 1)
```

Notou, 2022-10-12:
Hello,

Is the flowgraph you show the exact one you are trying to run? 
If so, the problem might be that the generate options are set to `QT GUI`, and not `Bokeh GUI`.

AllisonOge, 2022-10-12:
Thanks @Notou that solved it.

closed 2022-10-12 by Notou