gnuradio/gr-bokehgui #59: Gui config

utya, 2023-01-11:
Hello.
Can't find bokhe gui in dropdown menu
[image]

Notou, 2023-01-31:
Hi,
Please provide more information as to what you did to arrive to this issue, so we can help you.
What is your gnuradio version, what is the gr-bokehgui branch ?
Did you build gr-bokehgui, and where did you install it? Is it in a folder that gnuradio knows about?

ivandiazpardo, 2023-03-30:
Hi.
I have the same problem. Intall gnuradio version 3.10 ans install gr-bokehgui, refresh blocks and they appear.
make install put libraries en in folders gnuradio.
[image]

Notou, 2023-03-31:
Hi,
If I understand correctly, you installed and compiled gnuradio 3.10 and gr-bokehgui master?
The block yaml files seem to be installed in a sensible place.
The blocks show up in the GRC block list, but you don't have the "bokeh GUI" option in the dropdown menu?

Have you tried restarting GRC?

ivandiazpardo, 2023-03-31:
Hi,
Yes, I don't have the "bokeh GUI" option in the dropdown menu.

I retard GRC but the problem continue.

ivandiazpardo, 2023-03-31:
Ok, Notou.

I'll try. I haven´t run sudo ldconfig.

ivandiazpardo, 2023-03-31:
Hi,

Python interpreter show this message.
[image]

Notou, 2023-03-31:
This looks like you installed bokeh version >=3.
The WidgetBox has been removed.
It works with bokeh versions up to 2.4.3

ivandiazpardo, 2023-03-31:
So I have to remove bokeh 3.10 and install 2.4.3

Notou, 2023-03-31:
It's gnuradio that is 3.10
Bokeh versions go up to 3.2 I believe.

But yes, you need to downgrade the bokeh version

ivandiazpardo, 2023-03-31:
Hi,

Thanks, It's works. I installed bokeh version 2.4.3 with pip install bokeh==2.4.3 and now the bokeh GUI appears in the menu.
[image]

closed 2023-04-03 by Notou