gnuradio/gr-bokehgui #63: AttributeError: 'test_bokehgui' object has no attribute 'widget_lst'

johanEDSaputro, 2024-03-24:
i got some error due to running gr-bokehgui,

here is the errors
Traceback (most recent call last):
  File "/home/brawijaya/Documents/Documents-Thinkpad/Project/Research/GRC/plugins/gr-bokehgui/examples/./test_bokehgui.py", line 578, in <module>
    main()
  File "/home/brawijaya/Documents/Documents-Thinkpad/Project/Research/GRC/plugins/gr-bokehgui/examples/./test_bokehgui.py", line 556, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "/home/brawijaya/Documents/Documents-Thinkpad/Project/Research/GRC/plugins/gr-bokehgui/examples/./test_bokehgui.py", line 82, in __init__
    self.signal_amp_slider = bokehgui.slider(self.widget_lst, 'Signal Amplitude' +":", 0, 10, 0.01, 10, 10.0)
                                             ^^^^^^^^^^^^^^^
AttributeError: 'test_bokehgui' object has no attribute 'widget_lst'

any suggestion for fixing this problem? thank you

johanEDSaputro, 2024-03-25:
the shared library is not loaded, so i just need to running the grc from terminal

thanks

Notou, 2024-03-26:
Are you saying that you fixed your error?
Path issues in your environment variables could indeed happen when running GRC from a shortcut,
and could cause some modules to not be found.

johanEDSaputro, 2024-03-28:
yes the error was solved. The error occurred because the shared library not open in GRC shortcut, so I just need to running using terminal and I just edit my shortcut.

Thanks

> Are you saying that you fixed your error? Path issues in your environment variables could indeed happen when running GRC from a shortcut, and could cause some modules to not be found.

closed 2024-03-28 by johanEDSaputro