gnuradio/gr-bokehgui #69: BokehGui missing in Generate Options

r4d10n, 2024-06-25:
Environment:
```
	Arch Linux
	GNU Radio 3.10.7.0 
	Python 3.11.5
	Node 16.20.2
	Bokeh 2.4.3
```

I have installed gr-bokehgui as per instructions from github readme. Initially cmake completed even with `pygccxml` missing in my installation. I installed pygccxml and did `make, sudo make install and sudo ldconfig`. The BokehGui components are present among gnuradio-companion widgets. When I try to generate a flowgraph and execute it, it reports:

```
import bokehgui
ModuleNotFoundError: No module named 'bokehgui'
```
On checking web, I found that I hv to set the Generate Options inside the Options block to BokehGui, but that option is missing. 

When I tried running `make test` from the build directory, the log file showed the following error for all 7 files:

`ModuleNotFoundError: No module named 'bokehgui_swig'`

Kindly let me know how to go about this.

Notou, 2024-06-25:
A couple of things first:

1. The `make test` errors are "normal": it seems I've not updated the test files over the years, so they cannot work properly. It's not related to how well your installation worked.

1. Given the error message at import that you report, it's normal that you don't see a `Bokeh GUI` option in the Options block: GRC tries to import bokegui and only shows the option if the import succeeds.

Now the question is why does the import not work?
The dependencies versions you share seem good.
Are you working on the latest commit of the master branch?
Are you operating inside of a conda environment? 

Could you edit the python/__init__.py [file](https://github.com/gnuradio/gr-bokehgui/blob/bad6af7e3719a1999c3cbbc216ca49ae859ff44d/python/__init__.py#L33) to remove the try/except statement around `from .bokehgui_python import *` ?
This could give a bit more information as to where the error is.

r4d10n, 2024-06-25:
Thanks for the pointer... I am indeed using the master branch, with a non-conda environment.

My bad.. I had missed putting the new Python version's `/usr/local/lib/python3.x/site-packages` in PYTHONPATH. Adding that, `import bokehgui` worked and now it's all fine ! 

Cheers

closed 2024-06-25 by r4d10n